westlicht/performer #358: Routing: Transpose happens to late

bemtec, 2022-05-07:
When I route a transpose by a track CV, then the first step of the transposed track is most of the times not transposed yet. This happens also when I set the first track as the transpose track. Can you prioritise routings?

jgoney, 2022-10-26:
@westlicht I'm facing this as well it seems. I wonder if this is related to issue https://github.com/westlicht/performer/issues/167. I have track 1 routed internally to transpose tracks 2, 3, and 4. Everything works on time at 75 BPM and lower, but the transposition is about a 16th note or so late at any BPM higher than that. I know there are limitations with regards to processing power and this project's architecture, but is this something that could be reasonably addressed?

Solar-X, 2023-02-05:
Hi,
I tried the following - and it works:
I doubled the steps, set my sequence (has to be at most 32 steps) on step 2, 4, 6, 8, etc., put no gate at the new steps (1, 3, 5, 7, etc.) and doubled the speed of the sequence.
So the point of transposing comes on the empty step one and has already changed at the beginning of the real sequence  (step two).