westlicht/performer #385: Westlicht sd card

Sixdifferentways, 2024-03-23:
Hi! Just received yesterday my performer, naturally without sd card. These are my firsts steps with this module, so thanks in advice for your help…
1) which type of micro sd may I use?
2) how can I format my brand new micro sd? Fat32? Ntfs?
3) may I format my micro sd directly from my Performer? Can you write me the entire procedure?
4) Do I have to download and put in my sd card some file system or Performer will do it for me if I’ll format my card internally in him?

Thank you so much for your help: I think performer will became my best sequencer in few hours 😊

SloBloLabs, 2024-03-23:
16gb sd micro card will suffice. Format instructions can be found in the manual. Happy reading!

Sixdifferentways, 2024-03-23:
I tried yesterday to look in the manual but I couldn't find the instructions for formatting the card.

Sixdifferentways, 2024-03-23:
the question is: may I use a new sd card just to save my projects without downloading last update.dat file?

Sixdifferentways, 2024-03-23:
Is this attached the right procedure to format an sd card? It is named as FIRMWARE UPDATE
[image][image]